IoT-project-groups-1-2/g1-map-robot #27: Create PID for driving of motors based on Gyro Z-axis

vas-dav, 2022-11-23:
- Get current z-axis
- put it inside pid
- retreat direction from pid 
- set to motors

Redperij, 2022-12-04:
PID must be added to this function in order to determine motors speed depending on angle value.
```c++
//Pass angle_sum
uint8_t
try_to_correct (int ang_sum, uint8_t *cur_velocity_left, uint8_t *cur_velocity_right)
{
  if (ang_sum IS_LEFT)
  {
    if (*cur_velocity_right > 90) *cur_velocity_right -= 1;
    else if (*cur_velocity_left < 180) *cur_velocity_left += 1;
  }
  else if (ang_sum IS_RIGHT)
  {
    ang_sum *= -1;
    if (*cur_velocity_left > 90) *cur_velocity_left -= 1;
    else if (*cur_velocity_right < 180) *cur_velocity_right += 1;
  }
  SetMotors (0, 0, *cur_velocity_left, *cur_velocity_right, 0);
  return 0;
}

vas-dav, 2022-12-04:
Should the pid set to both motors or only one?

Redperij, 2022-12-04:
Should be both. You can change the logic of this function as long as it is not changing the angular speed too rapidly.
Remember, that this function is called very often, so the main problem is to choose proper speed to quickly return to the straight state, but not overdo it in order not to repeat the same process again, but from the other side.

closed 2022-12-09 by Redperij